eried/flipperzero-mayhem #9: proble with binaryies

grugnoymeme, 2023-07-26:
the version before, i used flash it throught the esptool updater of spacehuhn. but now every time i try to flash it it give me back: ffffxffxxxf  xfffx andd so... maybe the boot file, partition and bootloader not working anymore with this version?

grugnoymeme, 2023-07-26:
i tryed 5v or 3v or vcc but none of them seems to work

grugnoymeme, 2023-07-27:
Please Tell me if the process is right.
I do this way cause it was the only way that worked for me last time.
1.) Switch on the flipper.

2.) Connect the cam to flipper:
Cam: GND (not GND/RST), VCC, UOT, UOR
Flipper: GND (11), 3.3v(9) or 5v(1), RX, TX

3.) Put a bridge also between the pin I0o and GND (the near ones) on the cam.

4.) Connect the flipper to the pc (without opening qFlipper)

5.) Open GPIO/UART bridge (set COM 0) (Not the COM 1)

6.) Press "reset" button on the cam to enter bootloader mode

7.) Go to spacehuhn webupdate

8.) Upload the following files from YOUR repo (/docs/webinstaller/build), and set the following values near them:

- 1000 (esp32cam_marauder.ino.bootloader.bin)
- 8000 (esp32cam_marauder.ino.partitions.bin)
- E000 (boot_app0.ino)
- 10000 (the new firmware)(esp32cam_marauder.ino.bin)

9.) wait till the end of the flashing

10.) unplug the flipper from the pc

11.) close the UART bridge

12.) remove the I0o/gnd bridge on the cam

13.) push "reset" button on the cam (or unplug and replug)

14.) it should work...

NO :( fffffxffff bla bla bla.

eried, 2023-07-27:
Try with 

1) put a bridge io0 and GND
2) open gpio uart in the flipper and connect flipper to pc
3) connect esp32 with the bridge from 1)
4) press upload -> https://flipper.ried.cl/webinstaller/ select flipper
5) wait for process
6) remove bridge from 1) and reset

closed 2023-07-27 by eried

grugnoymeme, 2023-07-27:
no i tryed and its the same, but please can you tell me exctly what to do:

1 connect the cam (5v, 3.3v, or VCC??? cause 5v give me fxxf~~xxf~~ ecc, 3.3 and VCC is too slow or crash)
2 switch on the flipper
3 put a bridge between io0 and GND 
4 open gpio uart in flipper (in config. table on flipper should i select USB CHANNEL 0(CLI) or 1??)
5 connect flipper to pc (no qFlipper)
6 press upload -> https://flipper.ried.cl/webinstaller/
7 select flipper (which one?? cause if i selected USB CHANNEL 1, i have 2 serial connected of flipper)
8 wait for process
9 remove brige from io0 and GND (first, or after i unplugged the flipper from the pc?? and firs or after i exit the UART bridge???)

Thank u.. sorry but im going crazy litteraly ahah

eried, 2023-07-28:
Maybe is defective? in the webinstaller there is links to videos with the process uploading the fw, please repeat exactly what the videos show