eried/flipperzero-mayhem #11: Channel numbers on list -a

sfjuocekr, 2023-08-16:
The original ESP Marauder shows the channel numbers when you do a list -a, this helps tremendously when you need to know what channel to sniff on for PMKID's.

eried, 2023-08-17:
also in the flipper version? this is a fork of that project so it should behave the same 🙂

sfjuocekr, 2023-08-20:
Maybe the way it shows was changed, but when I use marauder on the WiFi devboard it does show me channels when I do a list -a.

There have been quite a few changes to marauder over the past weeks/months.

eried, 2023-08-21:
mmm, but when you say original marauder what do you mean? I do not really understand what changes you refer to. This is based on marauder esp32 so it should be the same behaviour unless something changed last weeks

sfjuocekr, 2023-08-30:
When I say original Marauder I meant pulled from https://github.com/justcallmekoko/ESP32Marauder/

I have since build my own from that repo with some changes and got it working like a bare bones marauder at the moment. It also seems to support BT!

eried, 2023-09-14:
Yes, it will be updated soon

closed 2023-09-14 by eried