eried/flipperzero-mayhem #12: Arduino sketch compilation errror

kyoaku, 2023-08-20:
Hi, I tried to compile the esp32cam_marauder in Arduino IDE with those dependencies : 

**Adafruit NeoPixel** version=1.11.0
**ArduinoJson** version=6.21.3
**AsyncTCP** version=1.1.4
**EloquentEsp32cam** version=1.1.2 (I saw a dependency to this lib in code)
**ESP Async WebServer** version=1.2.3
**JPEGDecoder** version=1.8.1
**LinkedList** version=1.3.3
**lv_arduino** version=3.0.1
**NimBLE-Arduino** version=1.4.1
**quirc** (the provided one)
**SwitchLib** version=1.1.1
**TFT_eSPI** version=2.5.0

And I have the following errors (related to LikedList lib)

```
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:29,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h:39:7: error: redeclared with 2 template parameters
 class LinkedList {
       ^~~~~~~~~~
In file included from E:\flipperzero-mayhem-master\esp32cam_marauder\configs.h:416,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\esp32cam_marauder.ino:8:
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: note: previous declaration 'template<class T> class LinkedList' used 1 template parameter
 class LinkedList{
       ^~~~~~~~~~
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:29,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h: In constructor 'StringArray::StringArray()':
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h:178:37: error: no matching function for call to 'LinkedList<String>::LinkedList(std::nullptr_t)'
   StringArray() : LinkedList(nullptr) {}
                                     ^
In file included from E:\flipperzero-mayhem-master\esp32cam_marauder\configs.h:416,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\esp32cam_marauder.ino:8:
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:181:1: note: candidate: 'LinkedList<T>::LinkedList(int, T) [with T = String]'
 LinkedList<T>::LinkedList(int sizeIndex, T _t){
 ^~~~~~~~~~~~~
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:181:1: note:   candidate expects 2 arguments, 1 provided
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:113:1: note: candidate: 'LinkedList<T>::LinkedList() [with T = String]'
 LinkedList<T>::LinkedList()
 ^~~~~~~~~~~~~
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:113:1: note:   candidate expects 0 arguments, 1 provided
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: note: candidate: 'constexpr LinkedList<String>::LinkedList(const LinkedList<String>&)'
 class LinkedList{
       ^~~~~~~~~~
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: note:   no known conversion for argument 1 from 'std::nullptr_t' to 'const LinkedList<String>&'
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:29,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h: In member function 'bool StringArray::containsIgnoreCase(const String&)':
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h:181:27: error: 'begin' was not declared in this scope
     for (const auto& s : *this) {
                           ^~~~
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h:181:27: note: suggested alternative:
In file included from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\string:51,
                 from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\bits\unique_ptr.h:37,
                 from c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\memory:80,
                 from C:\Users\Hola\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\FS\src/FS.h:24,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\configs.h:414,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\esp32cam_marauder.ino:8:
c:\users\hola\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\include\c++\8.4.0\bits\range_access.h:87:5: note:   'std::begin'
     begin(_Tp (&__arr)[_Nm])
     ^~~~~
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:29,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/StringArray.h:181:27: error: 'end' was not declared in this scope
[... 39 lines cut ...]
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:113:1: note:   candidate expects 0 arguments, 1 provided
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: note: candidate: 'constexpr LinkedList<AsyncWebHeader*>::LinkedList(const LinkedList<AsyncWebHeader*>&)'
 class LinkedList{
       ^~~~~~~~~~
c:\Users\Hola\Documents\Arduino\libraries\LinkedList/LinkedList.h:26:7: note:   no known conversion for argument 1 from 'DefaultHeaders::DefaultHeaders()::<lambda(AsyncWebHeader*)>' to 'const LinkedList<AsyncWebHeader*>&'
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:468,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/AsyncWebSocket.h: In member function 'bool AsyncWebSocketClient::canSend()':
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/AsyncWebSocket.h:229:43: error: 'class LinkedList<AsyncWebSocketMessage*>' has no member named 'length'
     bool canSend() { return _messageQueue.length() < WS_MAX_QUEUED_MESSAGES; }
                                           ^~~~~~
In file included from c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:469,
                 from E:\flipperzero-mayhem-master\esp32cam_marauder\EvilPortal.ino:3:
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/AsyncEventSource.h: In member function 'size_t AsyncEventSourceClient::packetsWaiting() const':
c:\Users\Hola\Documents\Arduino\libraries\ESP_Async_WebServer\src/AsyncEventSource.h:89:59: error: 'const class LinkedList<AsyncEventSourceMessage*>' has no member named 'length'
     size_t  packetsWaiting() const { return _messageQueue.length(); }
                                                           ^~~~~~

exit status 1

Compilation error: exit status 1
```
Did I miss a Arduino IDE configuration or anything else?

Thanks

eried, 2023-08-21:
Hi, in the folder of the code there is a different library for ESPAsyncWebServer, that uses an "alternative" name for the linkedlist :) try to use that one

kyoaku, 2023-08-26:
Hi, thanks, it's working

closed 2023-08-26 by kyoaku