eried/flipperzero-mayhem #13: Required lib seem to be missing

agentzex, 2023-08-31:
From the required libraries section
https://github.com/eried/flipperzero-mayhem/tree/master/esp32cam_marauder/libraries

Seems to be missing this lib:
https://github.com/eloquentarduino/EloquentEsp32cam

Without installing it from Arduino lib manager , I received compilation errors about missing #include in MotionDetection.ino

eried, 2023-09-14:
Hi, I think you solved this based on the emails you sent me

agentzex, 2023-09-14:
Yes I solved it eventually , but what I meant here is that it's missing from the libs folder in this repo  ( https://github.com/eried/flipperzero-mayhem/tree/master/esp32cam_marauder/libraries) although, at least for me, it was required for compilation

eried, 2023-09-15:
Oh, ok. I will include it. Thanks