eried/flipperzero-mayhem #17: v012.0 2023-09-14 firmware seems to just boot loop on Mayhem 1.1 board

rlaltrello, 2023-09-18:
So when I try the v0.12.0 2023-09-14 firmware, the ESP32-CAM just will stay in a boot loop.  Resetting, disconnecting, different micro sd card.  Nothing will allow it to work properly.  If I use the "esp32cam_marauder.ino.bin", from the history files (back in March), I can get the older version to at least boot properly.  

Interestingly, I can get the version 12 to load with the "vanilla" marauder flash (from and using FZEasyMarauderFlash) to boot just fine.  But it will not find the micro sd card through the ESP32-CAM -- but other functions that do not need the sd card work fine using the unleashed version [ESP32] Wifi Marauder app.

Any ideas why I cannot get your mayhem v1.1 hardware to flash with your files from your current webinstaller?

eried, 2023-09-18:
Can you see what the boot loop is saying? connect it via serial in Arduino IDE for example and 115200 or 230400. Paste here

rlaltrello, 2023-09-18:
Sure:

-----------
ELF file SHA256: f36ac39c7b69d770

E (520) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x40083da1:0x3ffe3ab0 0x4008effd:0x3ffe3ad0 0x400956c9:0x3ffe3af0 0x4010c0e2:0x3ffe3c20 0x400834b5:0x3ffe3c50 0x40079316:0x3ffe3c90 |<-CORRUPTED

-----------

Thanks!

eried, 2023-09-19:
Strange, based on 

```
E (520) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

I would say the browser failed to download the files and flashed something wrong. If you clear the browser cache and flash again I assume it would work. I have flashed many with the newest version without issues.

rlaltrello, 2023-09-19:
Switched to a windows 11 machine (was using MacOS before), same issue on a clean browser.   [This is what I am seeing.](https://youtu.be/ZKBJTPQxAn0)

eried, 2023-09-19:
I havent seen that before, I would try to upload the firmware using arduino directly and using the "Erase all flash" option in Tools while uploading it. Otherwise the board might be faulty

eried, 2023-09-25:
@rlaltrello have you figure out more about this boot loop?

rlaltrello, 2023-09-25:
Sorry for delay, life happened.  I have wiped the flash with the [ESP tool](https://espressif.github.io/esptool-js/) and tried again with the same results.  I guess it could simply be a bad board.

eried, 2023-11-08:
I will close this for now since is not replicable, but write back if you find out something else.