eried/flipperzero-mayhem #31: Serial Connection To ESP in wrong mode

MadPanda404, 2024-01-19:
Hey

When i connect my Flipper to usb and use any serial tool (screen, putty, and so on), input works as intended (i can edit my input and then hit enter so its get sent). But when i then switch the flipper in USB-to-UART, to use the marauder cli ion the mayhem v2 fin, i need to type really fast because input does not get send line by line, but key by key. is there a way to fix that?

eried, 2024-01-20:
is this in putty? or where

MadPanda404, 2024-01-21:
In putty, macos terminal, screen, i term... basically any terminal tool

eried, 2024-01-22:
I guess you set your bauds manually, but it should be 230400 for the ESP32 in this case -> check [this](https://github.com/eried/flipperzero-mayhem/wiki/Troubleshooting#the-apps-are-not-talking-with-the-esp32)

closed 2024-03-04 by eried