eried/flipperzero-mayhem #35: Suggestion/Addition

FatherDivine, 2024-02-29:
UHF RFID chip like the one found here : https://www.aliexpress.us/item/3256805110198094.html?spm=a2g0o.order_detail.order_detail_item.3.3c0af19cmyC0xL&gatewayAdapt=glo2usa

so tags like those found on asset tracking (and automatic car washes many times) can be read

eried, 2024-03-01:
oh thats a cool module, I am not sure how it could be integrated, maybe a new board?

FatherDivine, 2024-03-01:
Yes, that would be the main issue as the current Mayhem module is a nice, small size. But if somehow this chip could be incorporated all in one, I don't think there would be anything else worth adding. If not a possibility, it can always be another module.

FatherDivine, 2024-03-02:
> oh thats a cool module, I am not sure how it could be integrated, maybe a new board?

Shouldn't I be able to add that module to the extra PCB you provide on Tindie orders? I ordered a v2 yesterday, but don't have the UHF yet, but wouldn't mind working it onto the extra PCB. and just have an extra module.

eried, 2024-03-04:
Yeah, it could be dangling connected to v1, or a new board. I ordered one to test it out since it looks interesting. I will close this issue for now since is not fully related with mayhem or a problem

closed 2024-03-04 by eried