eried/flipperzero-mayhem #41: adding a gps module

runcom, 2024-05-23:
Hi, I'm trying to make a sense of how to add the gps module to the mayhem v2 board, is there any documentation about it? I can see the code mentioning the gps pins and such but wiring wise, I'm kind of lost

eried, 2024-05-23:
Technically you only need 1 pin for gps, the RX. If you arent using the microsd, you could hack your way out using those pins OR [gpio16](https://lastminuteengineers.com/esp32-cam-pinout-reference/), but I think that is on the back. Either way I am not sure how easy this will be but report how it goes. 

BTW the gps part in the firmware is because this is a fork, so if you use a normal ESP32 dev board there is a lot of free pins

runcom, 2024-05-23:
> Technically you only need 1 pin for gps, the RX. If you arent using the microsd, you could hack your way out using those pins OR [...]

what I'm basically trying to do is to put esp32-cam + NRF24l01 / CC1101 + gps (according to [this wiring](https://github.com/justcallmekoko/ESP32Marauder/wiki/gps-modification#connections)) to allow wardriving marauder too with the flipper zero. I basically would love to have [this](https://www.tindie.com/products/eried/mayhem-v2-for-flipper-all-in-1-espcamsdnrfcc/) plus the GPS.

My understanding is that the fz is connected to the esp32-cam but also the nrf24 and cc1101 are connected to it - but the GPS module has to be attached to the esp32-cam.

I'll try with just one pin as you suggested but I want to use the microsd (and in 1bit mode, I should have a couple of gpios available right?)

eried, 2024-05-23:
yeah, but I am not sure if they are exposed easily :S I dont remember in this moment if the gpio16 is actually in the back, I need to check later

runcom, 2024-05-23:
I read everywhere that gpio16 is actually busy with psram tho - so it might be unusable :/  maybe trying pin 12 or 13 may be better if the microsd works with only one pin

eried, 2024-05-23:
maybe, but I am not sure how easy is to use those pins, maybe you just need to test 1 by 1 which one is able to receive serial stream from the pin with SoftwareSerial(

runcom, 2024-06-02:
This change worked for me:

```
     #elif defined(MARAUDER_FLIPPER)
       #define GPS_SERIAL_INDEX 1
-      #define GPS_TX 9
-      #define GPS_RX 21
+      #define GPS_TX 12
+      #define GPS_RX 13
```
with proper connection to the esp32-cam :) happy to send a PR too 

```
...
Satellites: 7
  Accuracy: 3.00
  Latitude: redacted
 Longitude: redacted
  Altitude: 422.80
  Datetime: 2024-6-2 17:53:58
....
```

eried, 2024-06-03:
that would be great, specially adding the pinout :)