eried/flipperzero-mayhem #42: Firmware update? To latest marauder?

j0p3s, 2024-05-23:
Hi. Any idea when this will happen? Thanks in advance

eried, 2024-05-23:
oh, yes I think I should update :) maybe next weekend

j0p3s, 2024-05-23:
Thanks buddy. Have a great day

j0p3s, 2024-05-27:
what are the steps to compile it? Got a tutorial?

eried, 2024-05-28:
https://github.com/eried/flipperzero-mayhem/wiki/Compilation-of-the-firmware

But might be not 100% working since last version, I am still testing

j0p3s, 2024-05-30:
when compiling i get this error:

In file included from
C:\flipperzero-mayhem-next\esp32cam_marauder\EvilPortal.h:6,
                 from
C:\flipperzero-mayhem-next\esp32cam_marauder\esp32cam_marauder.ino:14:
C:\flipperzero-mayhem-next\esp32cam_marauder\ESPAsyncWebServer.h:31:10:
fatal error: StringArray.h: No such file or directory
   31 | #include "StringArray.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: StringArray.h: No such file or directory

j0p3s, 2024-05-30:
managed to fix that error but now i'm getting:

C:\flipperzero-mayhem-next\esp32cam_marauder\WiFiScan.cpp: In member
function 'void WiFiScan::executeSwiftpairSpam(EBLEPayloadType)':
C:\flipperzero-mayhem-next\esp32cam_marauder\WiFiScan.cpp:1911:5: error:
'esp_base_mac_addr_set' was not declared in this scope; did you mean
'esp_base_mac_addr'?
 1911 |     esp_base_mac_addr_set(macAddr);
      |     ^~~~~~~~~~~~~~~~~~~~~
      |     esp_base_mac_addr
Multiple libraries were found for "quirc.h"
  Used: C:\Users\Joao\Documents\Arduino\libraries\quirc_lib
  Not used: C:\Users\Joao\Documents\Arduino\libraries\ESP32QRCodeReader_Page
exit status 1

Compilation error: 'esp_base_mac_addr_set' was not declared in this scope;
did you mean 'esp_base_mac_addr'?

any ideas?

30/05/2024 à(s) 02:14):

eried, 2024-05-30:
Not sure, if you check https://github.com/justcallmekoko/ESP32Marauder/issues/418 seems to be also from marauder. I still havent finished working on the update. I need to update the wiki too

j0p3s, 2024-05-30:
> Not sure, if you check justcallmekoko/ESP32Marauder#418

Did add that line under line 22 and the error persists

eried, 2024-05-30:
OK, I need to compile myself first and ran tests, but I havent had time. If you solve it first just post here :)

j0p3s, 2024-05-31:
Have been trying to fix this issue but no luck whatsoever. I'm not a
programmer so i can't fix it myself but i'll keep looking

eried, 2024-06-04:
If I use this: https://github.com/eried/flipperzero-mayhem/wiki/Compilation-of-the-firmware it compiles OK 

The instructions changed a little but you can try again now. I havent tested on the board, but at least compiles

Inspirati, 2025-02-11:
simply add a forward declaration immediately before the function that calls it:

esp_err_t esp_base_mac_addr_set(uint8_t *Mac);