eried/flipperzero-mayhem #43: IRAM0 segment data does not fit.

runcom, 2024-05-24:
Seems like I've been able to compile everything but I get this at the end. I've been able to narrow the issue down to having PSRAM enabled in the board settings. If I disable it, it compiles, otherwise, it gives me this (notice I have no board attached yet, I'm just hitting "verify" in the arduino ide)

```
/Users/user1/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /private/var/folders/2q/b_hxf6nn50ddy8s9s8prfvfh0000gn/T/arduino/sketches/EF35CDE55DCEC7F8DF8D89DE5D8D5621/esp32cam_marauder.ino.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/Users/user1/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/Users/user1/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 1344 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

runcom, 2024-05-24:
> oh! could you try with a different Partition Scheme? without OTA for example, it is only 1.3 kb overflow 😂

same, it doesn't matter which partition scheme I use, if psram is enabled, I get this error :/
do you know why this is happening at all tho? 🤔

eried, 2024-05-24:
Not sure, because without psram I dont think the camera works. Is this the same code from this repo?

runcom, 2024-05-24:
> Not sure, because without psram I dont think the camera works. Is this the same code from this repo?

yup, unchanged code from this repo - I'm on osx if that changes anything, using v2.0.10 as ESP32 library in arduino ide 🤷‍♂️

runcom, 2024-05-24:
I think I've just bumped LinkedList from v1.3.1 to v1.3.3 and now it works.......no clue what happened

closed 2024-05-24 by runcom

eried, 2024-05-24:
oh, also I think esp32cam project now it is integrated to esp32, so I am not sure if now I should compile with the `AI Thinker ESP32-CAM` board instead. 

Eitherway, I need to update to latest marauder soon, so the code will change a bit.

runcom, 2024-05-24:
> Eitherway, I need to update to latest marauder soon, so the code will change a bit.

neat, I've actually already updated marauder in my fork and it seems to be working pretty well too :)

eried, 2024-05-24:
oh could you PR the update?

runcom, 2024-05-26:
> oh could you PR the update?

there you go https://github.com/eried/flipperzero-mayhem/pull/45