eried/flipperzero-mayhem #44: Mayhem Writing Empty Pcaps to Flipper SD

a1g0r1thm9, 2024-05-25:
I have installed and set up mayhem with my flipper running xtreme, and everything works well except that sniffed pcaps are not saved to the sd card. It saves logs, but only empty pcaps. My esp32-cam is some cheap thing i got from aliexpress, and seems to be an AI-Thinker clone.

eried, 2024-05-26:
Can you details the steps you do? does the photos get saved to your sd if you select that?

gianco75, 2025-04-19:
I have the same issue: the pcap files are empty on the Flipper's SD card, while on your device's SD card no files are being saved

eried, 2025-04-20:
Hi, are you sure the files are fully valid once you stop an operation in marauder? Can you detail the steps to create the empty pcaps?

gianco75, 2025-04-21:
*Hi, thanks for getting back to me.*
Here's what I do: I go to Apps > GPIO > Mayhem > Marauder > Scan, then I
select my AP in Select > Attack (deauth), wait a bit, then exit and quickly
go to Sniff (deauth).

Alternatively, I go to Apps > GPIO > Mayhem > Marauder > Scan, then select
my AP in Select > Sniff (pmkid) > Targeted Active (List), wait few minutes
and then exit.

Now, if I go to the folder apps_data > Marauder > Pcaps, all the files are
empty (on the Flipper's memory), and there are no files at all in the
Mayhem's memory.

eried, 2025-04-22:
OK, thanks I will update marauder during the week and give this a test

gianco75, 2025-04-22:
Ok thank you very much.
Bye

eried, 2025-04-29:
Hi! here is the new version, if you can test it I would appreciate feedback :) before making it public

https://flipper.ried.cl/webinstaller2/
https://flipper.ried.cl/webinstaller2/faps/mayhem_marauder.fap

gianco75, 2025-04-29:
Hi, thanks for your reply. I'll try it tonight and let you know

eried, 2025-04-30:
Awesome!