eried/flipperzero-mayhem #50: multiple definition of "text_table"

theundeadlich, 2024-06-09:
Followed the firmware compilation instructionswith the correct libs but keep getting this error when attempting to compile.

[image]

eried, 2024-06-10:
I think the files are duplicated? delete the whole folder and try again, or paste the log as text to check 🧐

closed 2024-07-06 by theundeadlich

theundeadlich, 2024-07-28:
Getting this error now:
"
(.text.ieee80211_raw_frame_sanity_check+0x5c): multiple definition of `ieee80211_raw_frame_sanity_check'; C:\Users\keetan\AppData\Local\Temp\arduino\sketches\27A501CE1AD08F8BA7710A48169C2420\sketch\objs.a(WiFiScan.cpp.o):C:\Users\keetan\Desktop\flipperzero-mayhem-next\flipperzero-mayhem-next\esp32cam_marauder/WiFiScan.cpp:13: first defined here
collect2.exe: error: ld returned 1 exit status
"

Followed the instructions in the FAQ to change the platform.txt file. https://github.com/justcallmekoko/ESP32Marauder/wiki/arduino-ide-setup#these-next-steps-only-apply-if-you-plan-to-build-the-full-esp32-marauder-firmware-from-source

and the note:
"In some versions of the platform.txt file from the marauder FAQ you wont find the line with compiler arguments, add -zmuldefs to the global compile flags, changing the line:

compiler.c.elf.libs="@{compiler.sdk.path}/flags/ld_libs"

with:

compiler.c.elf.libs=-zmuldefs "@{compiler.sdk.path}/flags/ld_libs"

I read in "https://github.com/justcallmekoko/ESP32Marauder/issues/181"
that I should "Do this except do it for whatever hardware version of the ESP32 you're working on."
so I wonder if i need the AI Thinker ESP32-CAM to be in the platform.txt file?

eried, 2024-07-29:
No, because the AI Thinker uses the ESP32 definitions (the ones that say only "esp32" without extras)