eried/flipperzero-mayhem #52: Eried can you please update this? 1.0.0 is already out and this is a few versions behind

j0p3s, 2024-09-02:
can you please update? 1.0.0 of marauder is out and the last update was almost a year ago... :(

j0p3s, 2024-09-02:
You told me in the last issue i opened you were going to update it and the time before and nothing

eried, 2024-09-02:
Yeah, I should. Haven't had time to fully test and release

j0p3s, 2024-09-02:
It would be cool if you could at least release the latest version. Would be
nice to test stuff with the v1 board ☺️. A big hug and thanks

eried, 2024-09-02:
I updated the source to the previous (before 1.0), but without testing much I do not want to release it. I can update the source to 1.0 but have you managed to compile the firmware yourself and test?

j0p3s, 2024-09-02:
I can try and compile it myself and test it out. I'll do it tomorrow and
I'll let ya know how it goes

eried, 2024-09-02:
You can test the latest version, the code is updated to 1.0 but you need to manage compiling it to test. I will ran my own tests in the future but I do not want to publish anything that could be buggy.

eried, 2024-09-11:
@j0p3s have you had a chance to test? I had to disable BLE functionality to make it fit :/ I am not sure how to optimize the original marauder to make it fit everything

j0p3s, 2025-04-19:
No I haven't tested it . Sorry

j0p3s, 2025-04-19:
I just lost hope that this will ever be updated

closed 2025-04-19 by j0p3s

eried, 2025-04-20:
It's updated in next, have you tested it? I'm not sure yet there is any benefit on updating yet

j0p3s, 2025-04-20:
Loads of things were implemented in a portal and so on. It is worth it .
They fixed several things

Com os melhores cumprimentos/ Without further ado,

João Pedro Rodrigues dos Santos

eried, 2025-04-22:
oh, ok, this week I will update it 🤠 I also need to fix the camera app not saving photos

eried, 2025-04-25:
still trying to update it, the original code is very sensitive to all the versions of libraries and boards :(

j0p3s, 2025-04-29:
> @j0p3s <https://github.com/j0p3s> Hi! here is the new version, if you can
> test it I would appreciate feedback :) before making it public
>
> https://flipper.ried.cl/webinstaller2/
> https://flipper.ried.cl/webinstaller2/faps/mayhem_marauder.fap11

I'll give ya feedback later today

Com os melhores cumprimentos/ Without further ado,

João Pedro Rodrigues dos Santos