eried/flipperzero-mayhem #54: Nanny Cam web site not loading, QR Code reader not working

aerialviews007, 2024-11-12:
Installed the release and the nanny cam web site does not load.  I also cannot get the QR code reader to pick up anything.  Are these functions dead?

eried, 2024-11-13:
No, they should work. What is the QR displaying? Does the camera works ok?

aerialviews007, 2024-11-13:
The QR FAP shows “waiting for QR code. 
the nannycam app shows the proper information for the SSID but the website doesn’t come up. I e tried multiple devices including adding http:// before. 
Camera, Morse code and Motion all work great though.

eried, 2024-11-13:
Since the resolution of the camera is very low, I suggest to you to try a bigger QR :) 

For the nannycam, could you try another phone? or maybe in your PC? I have seen reports of issues with some devices, mostly iphones.

aerialviews007, 2024-11-13:
Okay, I was able to get the QR code reader to work with a larger QR code.

Nannycam is not responding to HTTP requests.  I checked with wget.  "Read Error (Connection reset by peer) in headers"

eried, 2024-11-14:
Some browsers struggle with nannycam, but I haven't found the reason. It works on my android phone so if you can test and give me more info maybe I could fix it