eried/flipperzero-mayhem #55: "Triggered 300/4800 pixels" message after a few seconds

JMcrafter26, 2024-12-10:
Hi,

marauder and the camera are working, but when trying to connect to the Nanny Cam or using the QR Code scanner, I get this message:

```
Triggered 300/4800 pixels (threshold is 960)
```

(The number varies based on the changed pixels)

What is causing this, and how can I fix this?

eried, 2024-12-11:
Does it happen if you turn on more lights and have a plain background? can you post a photo here? (in marauder -> photo) and copy from the sd card?

1) it could be faulty camera module
2) it could be normal since the camera is quite low resolution so it has a lot of noise in low light situations 😬

JMcrafter26, 2024-12-11:
The thing is, it happens in any companion app (except camera and marauder) if I stay there for about 2 minutes. It is like this:
1. Open App
2. Module is working, app is working 
3. Wait a few minutes (sometimes less than one minute)
4. „Training“ message is display, followed by the triggered message
5. Need to restart the board

eried, 2024-12-11:
Do you see other chars before it happens? like `working, working... 2 mins -> invalid chars -> training -> triggered`?

I am asking because there is no retraining, unless 1) esp32 is faulty and reseting 2) battery on the flipper is too low 3) connection is faulty between both? (but should not happen if they dont experience movement or vibration)

https://github.com/eried/flipperzero-mayhem/blob/c3e0255331ad8fdd935c22ed36ec9799be960999/esp32cam_marauder/MotionDetection.ino#L47

if there is a fault it would be printed, but never "training" again

JMcrafter26, 2024-12-11:
Yes, sometimes there were characters like `~f~f` or something like that

I will try it again later, with a fully charged flipper