eried/flipperzero-mayhem #57: After fw flashing, marduar not working

Y8765, 2025-01-12:
Hi eried
Just flashed the board via you website, successfully. 
[image]
camera works fine, but seems marauder mayhem app, nanny cam, and QR Code aren't working, you're getting just a blank screen, or screen with some scramble letters and symbols.
and also the regular one that comes with momentum FW isn't working/scanning.
[image]
[image]
Would be happy to your advice.
thank you.

eried, 2025-01-12:
Hi, was it working before?

Y8765, 2025-01-12:
> Hi, was it working before? 

Most if it.
There were small problems with the maraduar from the beginning, but after fw flashing, I've been pwned :)

Y8765, 2025-01-13:
Maybe you can help me with newer compiled version, that will might help?

Y8765, 2025-01-13:
Can i erase the flash and rewrite it via the website?

Y8765, 2025-01-13:
I have flashed it multiple times, and still same result.
Not an hardware guy but hardware seems solid.
Its weird that the specific maraduer app isnt working, but for example, the mouse jacker which use the same antenna, works just fine.

eried, 2025-01-13:
It is not the same antenna, check the soldering, maybe one looks cracked? does the camera work?

Y8765, 2025-01-13:
And another weird fact, is that the subghz antenna, being detected on the flipper, only from one specific side, from the other side its not being detected.
Firmware update works from both sides of board.

Y8765, 2025-01-13:
camera works fine.
just maraduer, and nanny cam.
mores code works fine.

eried, 2025-01-13:
ah, that is strange then, because camera runs the same firmware. Then I think you arent opening the right marauder. Is the option that says [Mayhem] ? can you record a video of the issue showing how the camera works ok?

Y8765, 2025-01-13:
[image]
Nany cam flipper, stuck on this screen. worked before fw update.
QR Code stuck for 2 seconds, then gives the right output and working.
[image]

Y8765, 2025-01-13:
> ah, that is strange then, because camera runs the same firmware. Then I think you arent opening the right marauder. Is the option that says [Mayhem] ? can you record a video of the issue showing how the camera works ok?

yes, 2 min.

Y8765, 2025-01-13:
https://file.io/pITkhVtz1GIk
This video shows the apps functionality via qflipper, but because the apps requires detach before use, i have detached and reconnected after functionality started.
Can you upload some older firmware to your site for me to test it out?
My board is 2.0.3

Y8765, 2025-01-13:
If i make full chip erase via https://espressif.github.io/esptool-js/ and then rewrite it thrue your website, its fine?

Y8765, 2025-01-13:
Update: Even after full chip erase, and rewrite of latest firmware from site, still same problem.
Before we dive into fixing things, i want to try few older versions, but cant find any where to download.
Can you make a link in your website for some older verions?

eried, 2025-01-13:
Strange, but the QR code is not very accurate, needs a big and clear QR photo to work, a better test is the motion sensor. 

> Can you make a link in your website for some older verions?

Which version you need? All the versions are in github, but not on the web interface so I can write instructions how to recover an older version (i.e: [here](https://github.com/eried/flipperzero-mayhem/commits/next/docs/webinstaller/build) are all the changes, but you will need to use something like the esp32 uploader)

Y8765, 2025-01-13:
> Strange, but the QR code is not very accurate, needs a big and clear QR photo to work, a better test is the motion sensor.
>
>
> Which version you need? All the versions are in github, but not on the [...]

The motion sensor works fine.

Y8765, 2025-01-13:
> Strange, but the QR code is not very accurate, needs a big and clear QR photo to work, a better test is the motion sensor.
>
>
> Which version you need? All the versions are in github, but not on the [...]

I thought to test one or two fw before the latest version. 
In esp flasher programs I'm not quite sure which addresses to write each file.

eried, 2025-01-13:
The addresses are in the manifest, let me know how it goes. But if you print a big QR code that says "Hello" for example, it should work (like a full page QR, just to test)

```
{
  "name": "MayhemFirmware",
  "version": "release",
  "builds": [
    {
      "chipFamily": "ESP32",
      "improv": false,
      "parts": [
        { "path": "esp32cam_marauder.ino.bootloader.bin", "offset": 4096 },
        { "path": "esp32cam_marauder.ino.partitions.bin", "offset": 32768 },
        { "path": "boot_app0.bin", "offset": 57344 },
        { "path": "esp32cam_marauder.ino.bin", "offset": 65536 }
      ]
    }
  ]
}
```

Y8765, 2025-01-13:
> The addresses are in the manifest, let me know how it goes. But if you print a big QR code that says "Hello" for example, it should work (like a full page QR, just to test)
>
> ```
> {
>   "name": [...]

What the elf and .map are being used for?
Will update on progress.

Y8765, 2025-01-13:
> The addresses are in the manifest, let me know how it goes. But if you print a big QR code that says "Hello" for example, it should work (like a full page QR, just to test)
>
> ```
> {
>   "name": [...]

I flashed lower version of 13.5 r, all mayhem functions still not working.
QR Scan works fine
[image]

nany cam server dosent works

Y8765, 2025-01-13:
Not scans in reality, but app working back, but nothing more.
[image]

Y8765, 2025-01-13:
For conclusion, tried like 10 versions, including latest binaries from the repo, non of them solved the maraduer problem.
maybe you can try to release a new marduaer app?
IDK what to do, its a brand new board :(

Y8765, 2025-01-13:
of course everything flashed successfully.
[image]

Y8765, 2025-01-13:
Flashed successfully, and still
[image]

eried, 2025-01-13:
oh yeah maybe faulty board, seems wifi does not work. Please send a message via tindie support

closed 2025-01-13 by eried